QIDITECH/QIDI_SMART3 #1: home is not at the center

minus1ms, 2023-06-19:
i have X-Smart 3 and i updated firmware to 4.1.0, before it was 1.0.22

home is not moving head at the center, auto-leveling has same offset as home, it is slightly too close to left and front site.

minus1ms, 2023-06-19:
Also @Rainboooom can i have some private contact with you? It will be easier. Do u have discord for example? Im also programmer, i can always do some pull requests.

Rainboooom, 2023-06-20:
The reason should be that the `probe` parameter in the updated version of `printer.cfg` needs to be modified. There are currently two versions of smart probe positions, one on the left and one on the right.

We have a very responsible after-sales team, and you can directly send problems to them via email, and they will help you solve these problems.

The after-sales email address of the smart model are: Smart3ams@qd3dprinter.com / Smart3support@qd3dprinter.com

Our after-sales team will make every effort to help you solve the problem, and of course, you can also directly provide suggestions to our after-sales team.

minus1ms, 2023-06-20:
thx, i imagined that you are only one developer and after-sales team dont know about how it works and have to ask you anyways.

Rainboooom, 2023-06-20:
Never mind, it's my honor to help you.

closed 2023-06-20 by Rainboooom