QIDITECH/QIDI_SMART3 #2: movement beyond range of provision when auto leveling

minus1ms, 2023-06-20:
im getting "movement beyond range of provision" error when auto leveling, when printer is starting to collect 16 compensation points

minus1ms, 2023-06-20:
i guess that its something related to values in config too, im curious, is there like different variant of this printer? like why default values are wrong for my printer?

Rainboooom, 2023-06-21:
After the first batch of smart models were sold, we learned some issues from user feedback. We have made improvements to the hardware part of the machine. I remember I have already provided you with the contact information of the after-sales team before. You can inquire about relevant matters with them, and they will send you the new hardware facilities.

minus1ms, 2023-06-21:
alright, thx

closed 2023-06-21 by minus1ms

JojoDevel, 2023-07-02:
I am having similar issues with the firmware version 4.1.2 and contacted the support team