QIDITECH/QIDI_SMART3 #4: Malformed command 'SDCARD_PRINT_FILE FILENAME=XXXX'

donnydavis, 2023-10-24:
When trying to use Qidi Print, I receive this error 

Malformed command 'SDCARD_PRINT_FILE FILENAME=XXXXXXX'

For Instance here is the log output from klippy.log

Malformed command 'SDCARD_PRINT_FILE FILENAME=Ac Infinity Fan Bracket v1.gcode'

However, when I change the name to include underscores, this issue does not happen and the printer works properly.

Rainboooom, 2023-10-25:
Generally speaking, the naming of space characters can affect the reading of files, which is a normal phenomenon. Perhaps we should prompt not to include spaces in the name when naming slices.

donnydavis, 2023-10-25:
How can I test to see if whitespaces are being handled properly? The slicer is sending the file over properly, its just puking on the firmware side because I can assume whitespace has not been taken into account. 

Is there a method to get some deeper debugging enabled so I can run down where its puking at?

Rainboooom, 2023-10-30:
I will try working with colleagues in slicing software to see if I can solve this problem

QIDITECH, 2023-11-23:
We have fixed the bug, please update the firmware and software.

closed 2023-11-23 by QIDITECH