QIDITECH/QIDI_SMART3 #5: Optimise PRINT_START|END routines and move from Slicer to Klipper

andrewcharnley, 2023-10-26:
It's sub optimal to have the custom print start code into the slicer. It should be inside printer.cfg so it can be called by all slicer programs.

In addition the routine can be optimised to reduce moves and resolve leaky materials with the extruder by having an optional setting in the HOMING macro to prevent the head moving to the middle and then to build temperatures at the side of the bed. Once does the leaky extrusion will wipe on the side of the bed rather than sitting in the middle of the bed where the extruder will print on top!

The print start code can be optimised further to select a optimal exhaust fan based on the material. Slicers pass the material data as meta inside the gcode file. The exhaust fan shouldn't turn off immediately.

For certain materials the bed doubles up well as a chamber pre-heat. By passing the material this can be done automatically.

At the end of the print the head just immediately move to the corner, short blast the nozzle and lower the bed. This reduces and in most cases stops leaking materials from resting on top of the finished print.

The following code adds all the above functionality. It is well tested and in daily use.

* There's a chance I've missed some code so I've also attached the full printer.cfg.

`
[delayed_gcode AUTO_EXHAUST_SHUTDOWN]
gcode:
  SET_PIN PIN=exhaust_fan VALUE=0

[gcode_macro PRINT_START]
gcode:
    SET_PIN PIN=caselight VALUE=1
    RESPOND MSG='Material passed by slicer = {params.MATERIAL}'
   
    {% set BED_TEMP = params.BED_TEMP|default(50)|float %}
    {% set EXTRUDER_TEMP = params.EXTRUDER_TEMP|default(190)|float %}

    M140 S{BED_TEMP}
    G90

    G28 NO_CENTER=1 #home

    # Move the nozzle near the bed
    G1 X0 Y0 Z5 F3000
    # Move the nozzle very close to the bed
    # Any extrusion should catch on the side when the head moves
    G1 Z0.3 F300

    # Wait for bed to reach temperature
    M190 S{BED_TEMP}

    # chamber preheat
    {% set MATERIAL = params.MATERIAL|default('PLA')|string %}
    {% set CHAMBER_PREHEAT = 60 %} # seconds
    {% if MATERIAL == "ABS" %}
        G4 S{CHAMBER_PREHEAT} # wait for bed to heat chamber
    {% elif MATERIAL == "ASA" %}
        G4 S{CHAMBER_PREHEAT} # wait for bed to heat chamber
    {% endif %}

    # Set and wait for nozzle to reach temperature
    M109 S{EXTRUDER_TEMP}

    # exhaust fan
    UPDATE_DELAYED_GCODE ID=AUTO_EXHAUST_SHUTDOWN DURATION=0
    {% if MATERIAL == "PLA" %}
        SET_PIN PIN=exhaust_fan VALUE=255
    {% elif MATERIAL == "ABS" %}
        SET_PIN PIN=exhaust_fan VALUE=0
    {% elif MATERIAL == "ASA" %}
        SET_PIN PIN=exhaust_fan VALUE=0
    {% else %}
        SET_PIN PIN=exhaust_fan VALUE=127
    {% endif %}

[delayed_gcode DELAYED_PART_FAN_SHUTDOWN]
gcode:
  M106 P0 S0

[gcode_macro PRINT_END]
gcode:
    SET_IDLE_TIMEOUT TIMEOUT={printer.configfile.settings.idle_timeout.timeout}         ; set timeout back to configured value
    CLEAR_PAUSE

    # retract filament
    G1 E-3 F1800

    # extruder, bed cooldown
    M104 S0
    M140 S0

    # blast the extruder head to quicken the cooldown
    M106 P0

    # fans delay shutdown
    UPDATE_DELAYED_GCODE ID=DELAYED_PART_FAN_SHUTDOWN DURATION=6
    UPDATE_DELAYED_GCODE ID=AUTO_EXHAUST_SHUTDOWN DURATION=60
    
    # lower bed, move away from print
    G91
    G0 Z3 F600
    G90
    G0 X0 Y0 F3000
    G0 Z{printer.toolhead.axis_maximum.z-20} F600

    M220 S100 #feedrate
    M221 S100 #flow
    M84 # disable steppers

    BED_MESH_CLEAR
    BED_MESH_PROFILE LOAD=default
    G31

[output_pin exhaust_fan]
pin:PC9
pwm: True
cycle_time: 0.0100
hardware_pwm: false
value: 0
scale: 255
shutdown_value: 0.0

[homing_override]
set_position_X:0
set_position_y:0
set_position_z:0
axes:xy
gcode: 
    {% set HOME_CUR = 1 %}
    {% set driver_config = printer.configfile.settings['tmc2209 stepper_x'] %}
    {% set RUN_CUR = driver_config.run_current %}
    {% set HOLD_CUR = driver_config.hold_current %}

	{% if params.X is defined %}
	    SET_TMC_CURRENT STEPPER=stepper_x CURRENT={HOME_CUR * 0.8} 
        G28 X
		SET_TMC_CURRENT STEPPER=stepper_x CURRENT={HOME_CUR}
        G0 X10 F1200
    {% endif %}

    {% if params.Y is defined %}
		SET_TMC_CURRENT STEPPER=stepper_y CURRENT={HOME_CUR * 0.9} 
		G28 Y
		SET_TMC_CURRENT STEPPER=stepper_y CURRENT={HOME_CUR}
        G0 Y10 F1200
    {% endif %}

    {% if params.Z is defined %}
        G28 Z
        G0 Z0 F600  
    {% endif %}

    {% if params.X is undefined %}
    {% if params.Y is undefined %}
    {% if params.Z is undefined %}
    
    G91

    G28 Z
    G0 Z50 F600

    #G0 X10 Y10 F2400
    G4 P2000
    
    SET_TMC_CURRENT STEPPER=stepper_x CURRENT={HOME_CUR * 0.8} 
    G28 X
	SET_TMC_CURRENT STEPPER=stepper_x CURRENT={HOME_CUR} 

    # prevent part cooler clipping stepper driver mount
    G0 X10 F2400
    
	SET_TMC_CURRENT STEPPER=stepper_y CURRENT={HOME_CUR * 0.9} 
	G28 Y
	SET_TMC_CURRENT STEPPER=stepper_y CURRENT={HOME_CUR} 

    G90

    {% if params.NO_CENTER is undefined %}
        #G0 X{printer.toolhead.axis_maximum.x/2 - printer.probe["x_offset"]} Y{printer.toolhead.axis_maximum.y/2 - printer.probe["y_offset"]} F7800
        G0 X{printer.toolhead.axis_maximum.x/2} Y{printer.toolhead.axis_maximum.y/2} F7800
    {% endif %}

    {% endif %}
    {% endif %}
    {% endif %}
    SET_TMC_CURRENT STEPPER=stepper_x CURRENT={RUN_CUR} 
    SET_TMC_CURRENT STEPPER=stepper_y CURRENT={RUN_CUR} 

`

Rainboooom, 2023-10-30:
Thank you very much for your suggestion and configuration! I will immediately arrange a batch of machines for corresponding testing, and the things you mentioned are indeed very helpful to us!

andrewcharnley, 2023-10-30:
No problem. I also forgot, in PrusaSlicer (or alternative) use:

Start G Code

```
PRINT_START BED_TEMP=[first_layer_bed_temperature] EXTRUDER_TEMP=[first_layer_temperature] MATERIAL='[filament_type[initial_extruder]]'
```

End G Code
```
PRINT_END
```

closed 2023-11-23 by QIDITECH