QIDITECH/QIDI_SMART3 #6: Optimise Controller Fan Usage

andrewcharnley, 2023-10-26:
The controller fan is obnoxiously loud and irritating. It can be set to only come on when any stepper is active with an auto turn-off using the code below.

`[controller_fan controller_fan]
pin: PA8
kick_start_time: 0.5`

By default the fan is on all the time as it's installed in a +Vdd slot, but it can be moved to the unoccupied slot to the right for xsmart3. Adding this code for those that haven't moved it won't effect those printers.

Also note that in qidislicer it is including custom fan code for PA8, which is actually ignored for xsmart3. It's another reason why the qidislicer and klipper macros needs sorting!

Rainboooom, 2023-10-30:
I'm very sorry for the noise from the fan that made you irritable. Many users have indeed reported issues with fan noise, but we didn't think of this solution at the time. Your suggestions were very helpful to us, and we still need to work hard to optimize these configurations to give them better performance. Thank you again for your suggestions, and we are also very sorry for the bad experience you had.

andrewcharnley, 2023-10-30:
In my case I have replaced the controller fan with aluminium flat bar pieces and heat-tape on the CPU and Stepper heat-sinks, which then touch the steel metal base. In other words I have no controller fan, and CPU temperature drops 10%. I recommended to qidi email to look at spring copper pieces to do the same.

I fitted a Meanwell UHP-350-55, which is silent.

The exhaust fan I replaced with Orion 24V, they are the best and it's half the noise.

My X-smart is silent under no use and very quiet when under operation. For those without the mods though optimising the controller fan usage is an easy win.

QIDITECH, 2023-11-23:
Thank you for your feedback, we will optimize our products.

closed 2023-11-23 by QIDITECH

mirko-kienle, 2023-11-23:
@andrewcharnley thanks a lot for this as well as for the macros in your other issue. if you happen to have some handy, would you mind sharing pictures of the cooling setup you've described? i'm very tempted to try something similar myself. also do you have a part number for the orion fan you described? thanks and best

andrewcharnley, 2023-11-23:
Unfortunately I didn't take any before assembly but it's worth doing. The Meanwell UHP-350-55 fits well and the wires can be doubled over as they're closer, which almost resolves the flickering led on heating up. For the aluminium just buy some 1mm and 0.5mm thick bar and some 'led heat tape', both around 15mm wide.

The fan shouldn't be hard to find, they are also rebranded by rswww. Here's where I bought mine, the guy has hundreds. I can confirm it's at least half the noise, fabulous with the door shut! https://www.ebay.co.uk/itm/384796242826