QIDITECH/QIDI_SMART3 #7: ability to copy from machine to usb stick

insp1re1, 2023-12-12:
at present files uploaded to the machine cant be copied to usb stick for removal or backing up the machine files.

QIDITECH, 2023-12-13:
You can do this by connecting to the device over the network.
[image]

insp1re1, 2023-12-14:
> You can do this by connecting to the device over the network. [...]

this isnt what im asking for. what about in no network environments? all current market printers allow usb copy.

andrewcharnley, 2024-01-16:
I've never needed this in all my years of printing, what's the use case?

insp1re1, 2024-01-16:
What's it to do with you bozo.

andrewcharnley, 2024-01-16:
Well sometimes users ask for features without realising why. This smells like one of those times because data is flowing in one direction - you can't manipulate it on the printer and so it sounds like you're trying to do file management in the wrong place. This bozo is querying your use case as a sanity check.

insp1re1, 2024-01-16:
I know what I'm doing, secure environments don't all have network access.

andrewcharnley, 2024-01-16:
You're very quick to take offense! I don't have network access either. Come on then, describe your process.

insp1re1, 2024-01-16:
Nothing to do with you