QIDITECH/QIDI_SMART3 #10: Orca Slicer compatible print image

insp1re1, 2023-12-17:
Allow compatible printer preview images to be displayed when using orca, this should be a given as qidi sponsors orca, and is better than a black screen where the design should be.

QIDITECH, 2023-12-19:
We will change the preview image format to png in the next version, please wait for the update in the next version.

QidiPrinter, 2024-01-03:
> We will change the preview image format to png in the next version, please wait for the update in the next version.

slow basta, too slow

mirko-kienle, 2024-01-08:
The latest version of Orca Slicer ([1.9.0](https://github.com/SoftFever/OrcaSlicer/releases/tag/v1.9.0)) now automatically embeds preview images in the correct format, no need to wait for the firmware update. 🎉

closed 2024-01-10 by QIDITECH