QIDITECH/QIDI_SMART3 #11: Update UI failing?

digiblur, 2023-12-24:
The SOC updated to V4.1.12 but the UI seems to be stuck at 4.1.10.  Is there something I'm missing with the new update?  I've updated this printer before without issue.

FestivaMan, 2023-12-24:
I just went through this issue personally.  I emailed support and they had me try some other methods.  The one that ended up working I had to update the screen with a file they sent me directly with the sd card on the screen.  An email to support might be faster as they got back to me quickly on there

QIDITECH, 2023-12-25:
Please contact after-sales service to get the fix file, as the file cannot be sent on github.
https://qidi3d.com/pages/warranty-policy-after-sales-support

closed 2023-12-25 by QIDITECH