QIDITECH/QIDI_SMART3 #12: Unload filament

rednet1, 2024-01-01:
I installed 4.1.12 on x-smart 3, filament unload doesn't work.

regards

FestivaMan, 2024-01-01:
Did the UI update to 4.1.12 as well?

rednet1, 2024-01-02:
I tried the update again but only the SOC is at version 4.1.12, UI still 4.1.10 (mcu 0.10.0)
I did a reset but it doesn't change.

regards

FestivaMan, 2024-01-02:
Ok that's what was happening on both of mine as well. I ended up haveing to manually update the screen with an SD card.  Email the x-smart 3 support email that's on your printer and they will help you out with the file and instructions on how to do it.

rednet1, 2024-01-03:
Support gave me a file I think for the screen, but the simple update procedure doesn't work, I turn it off, wait 30 seconds, log in but it doesn't update, in the USB there is only the file they sent me.

FestivaMan, 2024-01-03:
Did you only try it with the USB port on the side of the machine?   Was the file something like MKSrecovery?  If so they my need to send you another file specifically for the screen, and there is a small SD card slot on the screen itself that I had to use to update my screen

rednet1, 2024-01-03:
Solved, 

I disassembled the display :)
Thank

closed 2024-01-03 by rednet1

FestivaMan, 2024-01-03:
Awesome.  Hope they can fix it for next time when it updates again.

michalvankodev, 2024-01-23:
This seems to be still an issue with the 4.1.13 version I have just put on my xsmart3. I'm sorry, but I don't have an idea how to resolve this issue from your conversation.

rednet1, 2024-01-23:
I also switched to version 4.1.13 which I just installed on my xsmart3 and I still had to disassemble the display.

weidi, 2024-01-24:
Same for me..., waiting for the email. 
@QIDITECH  wouldn´t it be great to document the procedure and files needed publicly ?