QIDITECH/QIDI_SMART3 #13: Ancient Interface

QidiPrinter, 2024-01-03:
Provide more modern interface, it looks like its from the 80s very poor. disgusting in fact.

FestivaMan, 2024-01-03:
I mean if you felt the need to make a profile just to come here and shit on the machines, that effort could have been put into building a Voron and you could choose whatever screen you wanted.  The screen works fine on mine, especially because it gets minimal use as most of everything gets done from the computer.

QidiPrinter, 2024-01-03:
Basta

FestivaMan, 2024-01-03:
Pasta?   You hangry or something?

QidiPrinter, 2024-01-03:
> Pasta? You hangry or something?

why you spam the thread?

FestivaMan, 2024-01-03:
Why you spam the repository with negative remarks about a printer that do absolutely nothing but complain?  If you would have made a suggestion about something like an adult you would get treated like one. But you're acting like a child so that's how you get treated.

QidiPrinter, 2024-01-03:
> Why you spam the repository with negative remarks about a printer that do absolutely nothing but complain? If you would have made a suggestion about something like an adult you would get treated like one. But you're acting like a child so that's how you get treated.

made suggestion of new interface!! You treat me no way motherless, its nothing to do with you.

QidiPrinter, 2024-01-03:
Your mother stinks

QidiPrinter, 2024-01-03:
> See. You're nothing but a child. Get a life and do something useful
> instead of being a burden to everyone around you. You're nothing but a joke
> to anyone that comes to look at this repository. Enjoy your pitiful,
> unfulfilled life.

Nobody knows who I am ya deadbeat. Make love to ya mother on the table bare
back bro

FestivaMan, 2024-01-03:
No one cares who you are either. There is ways to suggest things.  You're just a child that has no other way to communicate other than insults. It's comical. Maybe you should find a new hobby.

FestivaMan, 2024-01-03:
Why?  You looking for a new man?  Your current one isn't satisfying you enough that need need to come on GitHub and be rude to a company to get your rocks off?

QidiPrinter, 2024-01-03:
What you sleeping with men for? What's your obsession with qidi? Imagine
defending a Chinese printer. Climb on ya mother's back rent free bro

QidiPrinter, 2024-01-03:
> What's my obsession? You were the one that came here, and were ignorant
> and obnoxious. I just don't like people like you that think a company will
> do something if you treat them like garbage. So I figured you could use a
> taste of it yourself as I'm sure you're not used to it. And boy did you not
> disappoint. I'm having a grand time right now.
>
> I look forward to your childish answer shortly! I'll be here all day 😉

I'm laughing away, you ok with companies treating paying customers like
trash? Why not get a job bro? Then they can treat you like trash too.

Your mother's back is hairy...

FestivaMan, 2024-01-03:
I'm at my job getting paid right now.  

And please show me on the doll where Qidi hurt you.  Guess if you don't like them, maybe don't buy one?  Oh that's right you're just a little child so you haven't grown up to learn that yet 🤣. 

Maybe one day you will grow up.  But with all the insults you're throwing around I don't think it will be anytime soon. 

And with all the Mother insults it's getting concerning.  Did you mom not love you?  Did she hurt you?  It's ok to feel sad about that but trying to insult other peoples mother does nothing other than say your childhood was bad.  Hopefully you can get that figured out.

QidiPrinter, 2024-01-03:
Getting paid now too 😜😜😜😜 sodomising ya mother

FestivaMan, 2024-01-03:
See.  You must have been mistreated when you were little by her.  Because mom jokes have been out of style since the 90s.  But it's ok.  I kinda feel bad for you.  Not knowing what it was like to grow up in a nice family not being treated like crap.  That's why you feel the need to insult people you don't agree with.  I hope one day you will get the love you deserve (which isn't much from the looks of it, but I'm not completely cruel as even people like you need a little affection). 

I suggest you get out into the real world one day and see what it's really like.  Maybe you won't have to attack a Chinese company that's making affordable printers that work fine.  

And hopefully you can forgive your mom for giving you such a bad childhood that you never grew out of it.

QidiPrinter, 2024-01-03:
Pump away at your mother's shit box

QidiPrinter, 2024-01-03:
I'm 64 years old bro

FestivaMan, 2024-01-03:
I'm sure you are.....using words like bro and all.  It's ok. You can say whatever you like.  Age is a number.  Your mentality is less than a 7yr old. 64yrs old 🤣 your funny

QidiPrinter, 2024-01-03:
Age is a number, you say that to the kids you go meet till Chris Hansen
grips you?

FestivaMan, 2024-01-03:
That seems more like you area, as you were the one inquiring if I was gay, and talking about all that sort of things.    

And 64 years old 🤣.  I'm still laughing about that.  Anyone that age that says "bro" as much as you is a straight up joke of a human being.  No wonder you mom didn't love you. At this point I'm seeing what she must have saw 😉

QidiPrinter, 2024-01-03:
Must be of gay origin

FestivaMan, 2024-01-03:
Wow.  Original.  

You done with your nap then little guy?

QidiPrinter, 2024-01-03:
Gay?

FestivaMan, 2024-01-03:
Yea looks like you're running out of insults.  It's quite boring.  Get back to with something new "bro"

QidiPrinter, 2024-01-03:
U gay bro I need to know

QidiPrinter, 2024-01-04:
i was working, why so many messages? you gay bro?

FestivaMan, 2024-01-04:
Five hours to find a new insult.  And it's exactly the same.  Yea you're boring me.  Guess you're out of ideas and picking the same thing at the bottom of the barrel.  Pitiful really

QidiPrinter, 2024-01-04:
Well I had to go sleep with your mother then come back

FestivaMan, 2024-01-04:
Wow.  So original. Just like the last 10 times you said it.  You're like a toddler trying to insult someone in kindergarden. Enjoy your sad life alone.  You're boring me with the same insults.  You have got to be the lamest internet troll I have ever seen.  I'll go find someone that knows how to troll and can thing of original insults

QidiPrinter, 2024-01-04:
I'm not alone... I'm with your mother...

FestivaMan, 2024-01-04:
You're definitely alone.

QidiPrinter, 2024-01-04:
I have kids... I sleep with

QidiPrinter, 2024-01-04:
Fancy attempting anal intercourse? Use your mother's juices as lube

QIDITECH, 2024-01-04:
Thank you for your advice.

closed 2024-01-04 by QIDITECH