QIDITECH/QIDI_SMART3 #22: New filament unloading procedure wastes filament

erictelkkala, 2024-01-28:
The new way of unloading filament **assumes** that you can cut the filament just above the extruder, but X-smart 3 has the cutter below the extruder, not letting the gear move the cut filament back from the extruder to the plastic guide.

This requires you to cut the filament from the back of the machine before the run-out detector, wasting about 30 - 40 cm of filament to unload and load a new one.

Image of how much filament is wasted:
[image]

QIDITECH, 2024-01-29:
What is your firmware version? We have solved this problem in 4.1.13.

erictelkkala, 2024-01-29:
The firmware version is 4.1.13 for SOC and UI.
Firmware was updated yesterday.

QIDITECH, 2024-01-31:
I'm sorry, I misunderstood your meaning.
You can directly choose to unload the filament, rather than replace the filament, which will save a lot of materials.
[image]

antoineseveau, 2024-02-10:
Hello,
@QIDITECH ,where do you have this menu (screen) above in your post (on the SMART 3)? I just have choice to load or replace filament on 4.1.13 for SOC and UI.

rpg500e, 2024-02-17:
> Hello, @QIDITECH ,where do you have this menu (screen) above in your post (on the SMART 3)? I just have choice to load or replace filament on 4.1.13 for SOC and UI.

This UI only comes up during a print when the runout sensor is tripped and maybe when pressing pause during a print (not checked that yet). The issue you and I (and all) are having is this change to revert the up arrow was not done to the "load" section of the settings. 

@QIDITECH I appreciate the attention and support you guys are giving this firmware - would love to see the above change fixed

starbox82, 2024-02-18:
My brand new X-Smart 3 didn't even come with a filament cutter, just a space where it would be!  As the change filament procedure on the latest firmware release loads the filament then loads it again I am having to use M603 through fluidd to unload filament as it is incredibly wasteful to have to cut the filament at the end of the boden tube I have into the runout sensor (as my spool is to the side) and purge it through the system!

CChen616, 2024-02-21:
In the newest version 4.1.13, you can use up/down arrow button to unload/load filament, and the replace button will do a combination of both.

For those who are having issue with using up arrow to unload filament, please ensure you have updated your printer.cfg which is in inside the update package. Or you can manually check if the code below is inside your printer.cfg
```
[gcode_macro M603]
description: filament unlode
gcode:
    G92 E0
    G0  E100 F400

[gcode_macro M604]
description: filament lode
gcode:
    M83
    G1  E80 F400
```

starbox82, 2024-02-21:
> In the newest version 4.1.13, you can use up/down arrow button to unload/load filament, and the replace button will do a combination of both.
>
> For those who are having issue with using up arrow to [...]
There is no “up” arrow on my system (screenshot attached).  Changing the gcode_macro M603 does not change the behaviour of the change filament button on my system, it only works if I call it from the fluidd interface on my laptop.

The original code post firmware update is:

[gcode_macro M603]
description: filament unlode:
    G92 E0
    G0  E15 F400
    G4  P1000
    G92 E0
    G0  E-80 F800

But the screen doesn’t seem to pay any attention to that either.  I tried resetting to factory settings and changing the gcode again but have the same result as before.
[image]
[image]

CChen616, 2024-02-21:
> > In the newest version 4.1.13, you can use up/down arrow button to unload/load filament, and the replace button will do a combination of both.
> > For those who are having issue with using up arrow to unload filament, please ensure you have updated your printer.cfg which is in inside the update package. Or you can manually check if the code below is inside your printer.cfg
> > ```
> > [gcode_macro M603]
> > description: filament unlode
> > gcode:
> >     G92 E0
> >     G0  E100 F400
> > 
> > [gcode_macro M604]
> > description: filament lode
> > gcode:
> >     M83
> >     G1  E80 F400
> > ```
> 
> There is no “up” arrow on my system (screenshot attached). Changing the gcode_macro M603 does not change the behaviour of the change filament button on my system, it only works if I call it from the fluidd interface on my laptop.
> 
> The original code post firmware update is:
> 
> [gcode_macro M603] description: filament unlode: gcode: G92 E0 G0 E15 F400 G4 P1000 G92 E0 G0 E-80 F800
> 
> But the screen doesn’t seem to pay any attention to that either. I tried resetting to factory settings and changing the gcode again but have the same result as before. ![IMG_2837](https://private-user-images.githubusercontent.com/110785164/306544890-b529082a-6760-4dd0-99f5-9e8863ce3b7d.jpeg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDg1MDQzMjQsIm5iZiI6MTcwODUwNDAyNCwicGF0aCI6Ii8xMTA3ODUxNjQvMzA2NTQ0ODkwLWI1MjkwODJhLTY3NjAtNGRkMC05OWY1LTllODg2M2NlM2I3ZC5qcGVnP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDIyMSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDAyMjFUMDgyNzA0WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9MzM2ODg1Y2M3Y2RkNzhmMTE2ZWFhNGJiZWJhNGVjMzFiY2ZiZWRmZjU0MjA4NmVjY2FkZTA5NzcyNjAzODMwMCZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.7xig7sZJpARppfxqXAdGTnl9AdBZaFiA7p7fxPdU9Gw) ![IMG_2836](https://private-user-images.githubusercontent.com/110785164/306544922-f1515fc7-ed78-4eb6-b819-ed50bc38d949.jpeg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDg1MDQzMjQsIm5iZiI6MTcwODUwNDAyNCwicGF0aCI6Ii8xMTA3ODUxNjQvMzA2NTQ0OTIyLWYxNTE1ZmM3LWVkNzgtNGViNi1iODE5LWVkNTBiYzM4ZDk0OS5qcGVnP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDIyMSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDAyMjFUMDgyNzA0WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9ZGI4NzA1OGI2MjVmY2I4YzcyYjVjYjQ4YzkzNmNlMzU4OWFjOTIwMzIzMjExY2ZmNjJlZGE1NGVjY2I0MTIzMCZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.R7XTGZTCWRrTkJ9hZdk_I16eaJkMapzoEkrHQGOVuMA)

Thank you for your feedback. Currently, the up and down arrows are only available on the print pause interface for smart models. For filament unload when not printing, we recommend using fluidd to enter gcode commands (M603) or opening the machine's top cover to pull out the bowden tube and cut the end of the material. 

In this version replace filament button is calling M604, that's why changing the gcode_macro M603 does not change the behaviour of the change filament button.

We plan to fix this issue in the next version and apologize for any inconvenience caused.

starbox82, 2024-02-22:
Thank you CChen616.

robinvoor, 2024-05-02:
What is the ETA for this update?

CChen616, 2024-05-02:
The team hasn't provided a specific date yet, as the schedule is currently quite full. However, I personally anticipate it will be by the end of this month.

antoineseveau, 2024-06-21:
> The team hasn't provided a specific date yet, as the schedule is currently quite full. However, I personally anticipate it will be by the end of this month.

It’s a very long month :(

CChen616, 2024-06-21:
The online update, supporting for QIDI Link and other features requires a security upgrade, so the plan has been postponed. 😢