QIDITECH/QIDI_SMART3 #25: Homing can damage printer

minus1ms, 2024-01-29:
When homing, the head of the printer is moving a bit back and right. When head is already at the back (customer can put it here for example when cleaning), it will forcefully push back and can cause damage. I suggest removing this back movement. I have no idea why it would be needed.

CChen616, 2024-01-30:
The initial movement you've observed, where the print head moves slightly back and to the right, is actually a necessary step in the homing process. 

This slight retraction movement prevents potential false readings that could happen if it started from a position where it's already pressing against the endstops,  for example the front-left corner. If the print head is already at the back, it's best to manually move it slightly forward.

I hope this clarifies the importance of the movement you described. Please feel free to reach out if you have any more questions or concerns.

closed 2024-01-31 by CChen616

minus1ms, 2024-01-31:
the thing is that when it goes to front, its not damaging anything, it stops as soon as it feels resistance, however when it goes back, it does that with full power and in the worst scenerio could break the belts, cant you do some resistance check when it goes back? so it goes back normally as always but if its too close to the back then it will feel resistance and will stop and then it can go to front knowing that its already at the back so it wont create a new problem

minus1ms, 2024-01-31:
@CChen616 im pinging to make you see that