QIDITECH/QIDI_SMART3 #29: Set X Offset Not Woking After Saving

axim2661, 2024-02-09:
I can complete the Z offset (min is -1.31), the x-smart will save the settings. 
When I go print it appears to use the standard 0 offset.
Machine has been turned on and off and firmware reset.

minus1ms, 2024-02-17:
Hi, i have the same issue

CChen616, 2024-02-19:
Try to do a factory reset first. If it doesn't work, try to reinstalling the newest update package.

minus1ms, 2024-03-03:
Its not helping.

axim2661, 2024-03-03:
After completing the bed leveling and input shaping, I was able get the z offset to save correctly.

CChen616, 2024-03-04:
In the older version, when performing bed leveling, the last saved Z-offset value would be displayed. However, in the new version, it will show as 0. In fact, when you perform bed leveling, and the machine prompts you to adjust up or down, the Z-offset is already at 0, which might lead to confusion.

If the issue is not caused by the above reason, please attach the `/home/mks/klipper_config/config.mksini` file and the `/home/mks/klipper_logs/klippy.log` file for troubleshooting.