QIDITECH/QIDI_SMART3 #30: XY homing can damage printer

minus1ms, 2024-02-18:
Hi, XY homing can damage printer if head is already at the back. This happens because in printer.cfg in [homing_override] there are gcodes that are doing it.
```
    G91
    G1 X10 F2400
    G1 Y10 F2400
    G4 P2000
```
These gcodes are needed because without them there can be a problem at the front of the printer (homing wont be always accurate, it can bounce off the wrong element in the printer and cause a slight shift which could make bed mesh misleading).

I have figured out the safe way to do that. It is adding a bit more movement, but you have to sacrifice something. I have replaced the gcodes that I have mentioned above with this:
```
    G91

    # make sure that X is in the safe place
    SET_TMC_CURRENT STEPPER=stepper_x CURRENT={HOME_CUR * 0.8} 
    G28 X
	SET_TMC_CURRENT STEPPER=stepper_x CURRENT={HOME_CUR}
    G1 X50 F2400

    # make sure that Y is in the safe place
    SET_TMC_CURRENT STEPPER=stepper_y CURRENT={HOME_CUR * 0.9} 
    G28 Y
    SET_TMC_CURRENT STEPPER=stepper_y CURRENT={HOME_CUR}  
    G1 Y50 F2400

    # ready for the homing
```

I needed to use homing macros in here NOT to home BUT to prepare everything BEFORE homing.
It makes sure that homing XY will NOT damage anything in the printer and WILL BE 100% accurate.
This is the entire [homing_override] section after I have made the changes (in case you are unsure):

```
[homing_override]
set_position_X:0
set_position_y:0
set_position_z:0
axes:xy
gcode: 
    {% set HOME_CUR = 1 %}
    {% set driver_config = printer.configfile.settings['tmc2209 stepper_x'] %}
    {% set RUN_CUR = driver_config.run_current %}
    {% set HOLD_CUR = driver_config.hold_current %}

	{% if params.X is defined %}
	    SET_TMC_CURRENT STEPPER=stepper_x CURRENT={HOME_CUR * 0.8} 
        G28 X
		SET_TMC_CURRENT STEPPER=stepper_x CURRENT={HOME_CUR}     
        BEEP I=1 DUR=100       
        G1 X10 F1200
    {% endif %}

    {% if params.Y is defined %}
		SET_TMC_CURRENT STEPPER=stepper_y CURRENT={HOME_CUR * 0.9} 
		G28 Y
		SET_TMC_CURRENT STEPPER=stepper_y CURRENT={HOME_CUR}  
        BEEP I=1 DUR=100          
        G1 Y10 F1200
    {% endif %}

    {% if params.Z is defined %}
#        G90
#        G1 X{printer.toolhead.axis_maximum.x/2 - printer.probe["x_offset"]} Y{printer.toolhead.axis_maximum.y/2 - printer.probe["y_offset"]} F7800
#        G91

        G28 Z
        G1 Z20 F480     
        BEEP I=1 DUR=100	  
    {% endif %}

    {% if params.X is undefined %}
    {% if params.Y is undefined %}
    {% if params.Z is undefined %}

    G91

    # make sure that X is in the safe place
    SET_TMC_CURRENT STEPPER=stepper_x CURRENT={HOME_CUR * 0.8} 
    G28 X
	SET_TMC_CURRENT STEPPER=stepper_x CURRENT={HOME_CUR}
    G1 X50 F2400

    # make sure that Y is in the safe place
    SET_TMC_CURRENT STEPPER=stepper_y CURRENT={HOME_CUR * 0.9} 
    G28 Y
    SET_TMC_CURRENT STEPPER=stepper_y CURRENT={HOME_CUR}  
    G1 Y50 F2400

    # ready for the homing
    
    SET_TMC_CURRENT STEPPER=stepper_x CURRENT={HOME_CUR * 0.8} 
    G28 X
	SET_TMC_CURRENT STEPPER=stepper_x CURRENT={HOME_CUR} 
    BEEP I=1 DUR=100  
    G1 X10 F1200
    
	SET_TMC_CURRENT STEPPER=stepper_y CURRENT={HOME_CUR * 0.9} 
	G28 Y
	SET_TMC_CURRENT STEPPER=stepper_y CURRENT={HOME_CUR} 
    BEEP I=1 DUR=100        
    
    G90
    G1 X{printer.toolhead.axis_maximum.x/2 - printer.probe["x_offset"]} Y{printer.toolhead.axis_maximum.y/2 - printer.probe["y_offset"]} F7800
    G91

    G28 Z
    BEEP I=1 DUR=100  
    G1 Z50  F480
    {% endif %}
    {% endif %}
    {% endif %}
    SET_TMC_CURRENT STEPPER=stepper_x CURRENT={RUN_CUR} 
    SET_TMC_CURRENT STEPPER=stepper_y CURRENT={RUN_CUR} 
    M204 S10000

    G90
#    G1 X0 Y0 F7800
```

It will be good if you fix the issue in your printer.cfg. I will be thankful and I think that other customers will benefit from it too. I see some cracks on my printer's parts and I think that it was because of this issue.

CChen616, 2024-02-21:
Thank you for your suggestion, we will take this into consideration.

minus1ms, 2024-04-17:
Hi, i have figured out that TMCs are misconfigured and thats why this needs custom homing that is very long, i will try to configure that correctly