QIDITECH/QIDI_SMART3 #32: The printer wont start if the name of the file contains "é" characters

antoineseveau, 2024-03-06:
If the name of the body contains a "é" in QIDISLICER,  accent character are very commun in french language, the printer wont start, the sending to the printer with wifi seems to works, but then nothing happens, no start print
[image]
thanks

CChen616, 2024-03-07:
If possible, please attach a copy of the G-code for us to use for testing.

antoineseveau, 2024-03-07:
Of course,
the g-code name changes itself when i saved it. the "é" change in "e".
Here the g-code for this part :
[image]

[(Non enregistre).gcode.zip](https://github.com/QIDITECH/QIDI_SMART3/files/14520765/Non.enregistre.gcode.zip)

CChen616, 2024-03-14:
We have tested the generation of G-code files with French characters in their names on both Windows and iOS platforms and sent them to the printer for printing. Currently, there are no issues observed. Although spaces in the file names are converted to underscores and 'é' is converted to 'e', it seems to not affect the printing process. 

If the problem reoccurs, please attach the relevant information displayed in the Fluidd console along with klippy.log and moonraker.log for our analysis.

antoineseveau, 2024-03-14:
MOONRAKER LOG
2022-12-16 22:44:53,854 [shell_command.py:_check_proc_success()] - Command (/home/mks/moonraker-env/bin/python /home/mks/moonraker/moonraker/components/file_manager/metadata.py -p /home/mks/gcode_files -f ".cache/test e.gcode") successfully finished
2022-12-16 22:45:15,618 [websockets.py:on_close()] - Websocket Closed: ID: 281473392980320 Close Code: 1001, Close Reason: None, Pong Time Elapsed: 9.25
2022-12-16 22:45:16,377 [app.py:log_request()] - 101 GET /websocket?token=CGOKPHSTUUHHIAMLNK3OJQ2RCPMF57SC (192.168.1.18) [_TRUSTED_USER_] 3.82ms
2022-12-16 22:45:16,378 [websockets.py:open()] - Websocket Opened: ID: 281473392443008, Proxied: True, User Agent: Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/121.0.0.0 Safari/537.36, Host Name: 192.168.1.24
2022-12-16 22:45:16,389 [websockets.py:_handle_identify()] - Websocket 281473392443008 Client Identified - Name: fluidd, Version: 1.19.0-6184c73, Type: web

KLIPPY LOG : 
 Traceback (most recent call last):
  File "/home/mks/klipper/klippy/extras/virtual_sdcard.py", line 264, in work_handler
    self.gcode.run_script(line)
  File "/home/mks/klipper/klippy/gcode.py", line 216, in run_script
    self._process_commands(script.split('\n'), need_ack=False)
  File "/home/mks/klipper/klippy/gcode.py", line 200, in _process_commands
    self._respond_error(str(e))
UnicodeEncodeError: 'ascii' codec can't encode character u'\xe9' in position 51: ordinal not in range(128)
Exiting SD card print (position 361)

[moonraker.log](https://github.com/QIDITECH/QIDI_SMART3/files/14600030/moonraker.log)

CChen616, 2024-03-15:
We are still unable to reproduce this issue on our machines. Similar to what was reported in this issue, it seems to be a pre-existing problem with the official Klipper firmware.

https://github.com/Klipper3d/klipper/issues/4794

The current recommendation is to avoid including unusual characters in G-code file names and file contents (for example, content that might be auto-generated by slicing software), and to use UTF-8 encoding.

closed 2024-03-22 by CChen616