QIDITECH/QIDI_SMART3 #33: Printer EOL?

insp1re1, 2024-05-20:
no updates or fixes for ages now.

CChen616, 2024-05-20:
The next update is expected before mid-next month.

MuldersMakes, 2024-06-25:
guess you made that up?

CChen616, 2024-07-10:
Support related to QIDIlink requires security updates, and there have been personnel changes, so it will take some time to adjust.

theclue, 2024-11-10:
I must say that going EOL so quickly was incredibly disappointing to me. Something to carefully consider if planning to buy again...

QIDITECH, 2024-11-11:
Because we have launched a new product, so stop the production of the old model, maybe you can choose to buy the Q1 Pro or other models, they are better, and more stable, although it may be more expensive, but it is worth it.

andyhopkins, 2024-12-28:
Are you saying that the X-Smart 3 is now end of life and that no more firmware updates will be released?

This is very disappointing if so, and it has put me off of purchasing another Qidi device. I purchased the X-Smart because of its small footprint on the desk and that it was an 'appliance' rather than a tinker project. I feel that I would need to go to one of the more commercial companies (Bambu or Creality) for my next device, as they would support for longer.

MuldersMakes, 2024-12-28:
Yeah EOL, I sold all my qidi machines, the firmware is terrible.

Phil1988, 2025-01-24:
@MuldersMakes seems like you missed [FreeDi](https://github.com/Phil1988/FreeDi)

maybe also for you @andyhopkins and @insp1re1 as @theclue is already part of the familiy

MuldersMakes, 2025-01-24:
Already disposed of the printer

Phil1988, 2025-01-24:
Both original/stock WiFi dongles are fully supported on FreeDi

sgghostrider, 2025-01-24:
> Both original/stock WiFi dongles are fully supported on FreeDi

Nice, thanks for letting me know, then i will flash it as soon as i can, wifi is nice to have, specially when my ethernet cable has the tab broken and keeps getting disconnected every now and then...😅

Phil1988, 2025-01-24:
Just make sure you have a wired connection while first installation

andyhopkins, 2025-01-24:
Sounds really interesting I will take a look

starbox82, 2025-04-18:
Amazing that there's an open version of firmware for these printers - thank you @Phil1988 and everyone involved.  

Shame on @QIDITECH for abandoning its users so soon after bringing this product to market.  Telling people to buy your new printer is not customer support (besides, it seems they catch fire....).  I really liked the idea of the X-Smart 3, but given how quickly firmware updates have been abandoned I doubt I'll be buying a Qidi when I upgrade later this year.  

Hopefully FreeDi will fix the bugs Qidi didn't bother to as well as the bizarre filament changing process inflicted on us by the last official update!

Phil1988, 2025-04-22:
> Hopefully FreeDi will fix the bugs Qidi didn't bother

I dont know any stock bug that FreeDi hasnt already wiped out long ago (see my issues section). 
If you know one, please name it and I'll fix it ;)