QIDITECH/QIDI_SMART3 #37: Time setting

Mike3987, 2025-01-21:
The time is incorrect. Was wondering why ntp is not installed default in the firmware and user have to install manually.

QIDITECH, 2025-01-23:
You can refer to the following links to calibrate the time.
https://wiki.qidi3d.com/en/Memo/System-Time-Modification

Mike3987, 2025-01-23:
Just wondering why internet time is not integrated into the firmware?

sgghostrider, 2025-01-23:
The "time fix guide" is missing an important point, if the time is wrong(which it is from factory, mine was set to like several years in the past) installing anything gives an error message saying "certificate verification failed" because of the repository and local time mismatch. With method 2 can be fixed temporarily, and THEN you can install ntp, but for me it was not working for some reason, on every restart it was getting reverted to the old time.

The truth is even the latest firmware is outdated by years anyway(OS, klipper, web interface, everything), so instead of headbutting the wall i recommend to anybody who doesn't just "stick usb in hole -> select file -> print -> repeat" to install one of the alternative firmwares available, FreeQIDI works quite well, but you need to buy an "EMMC adapter" to be able to flash the storage. Only problem is the wifi driver for the dongle doesn't work out of the box, so you gotta buy a new compatible dongle, compile the driver or just use the ethernet cable, for me by cable works just fine. But just be aware that this voids the warranty, so do it once you are tired of breaking walls in your home out of frustration ;)